jcurl/RJCP.DLL.SerialPortStream #153: Linux Port Open Fails

abdullahbaa5, 2025-03-04:
Linux: Debian 12
SerialPortLib Version: 3.0.1

Build and copied libnserial (from the provided cmake code in this repo):
[image]

Error:
```
Unexpected baudrate returned after set (0; Input/output error)
   at RJCP.IO.Ports.Serial.UnixNativeSerial.ThrowException()
   at RJCP.IO.Ports.Serial.UnixNativeSerial.SetPortSettings()
   at RJCP.IO.Ports.SerialPortStream.Open(Boolean setCommState)
   at RJCP.IO.Ports.SerialPortStream.Open()
```
[image]

Usage Code:
```
_serialPort = new SerialPortStream();
_serialPort.PortName = _settings.AtPort;
_serialPort.BaudRate = 115200;
_serialPort.Parity = Parity.None;
_serialPort.DataBits = 8;
_serialPort.StopBits = StopBits.One;
_serialPort.Handshake = Handshake.None;
```

jcurl, 2025-03-04:
This occurs at line https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/f118746af66d9dea2871adc97aef887acd20185d/dll/serialunix/libnserial/openserial.c#L412

I don't know about the serial port or driver, but it's saying that a different baud rate is set than the 115200 that you've set. You can try commenting out that code, or enable logging of the library itself to get more information.

abdullahbaa5, 2025-03-05:
> This occurs at line
> 
> [RJCP.DLL.SerialPortStream/dll/serialunix/libnserial/openserial.c](https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/f118746af66d9dea2871adc97aef887acd20185d/dll/serialunix/libnserial/openserial.c#L412)
> 
> Line 412 in [f118746](/jcurl/RJCP.DLL.SerialPortStream/commit/f118746af66d9dea2871adc97aef887acd20185d)
> 
>  serial_seterror(handle, ERRMSG_UNEXPECTEDBAUDRATE); 
> I don't know about the serial port or driver, but it's saying that a different baud rate is set than the 115200 that you've set. You can try commenting out that code, or enable logging of the library itself to get more information.

I enabled logging and here are hte logs:
```
2025.03.05-04:44:22.479995      6       /dev/ttyUSB2    open: succeeded
2025.03.05-04:44:22.480252      7       /dev/ttyUSB2    setproperties: getting attributes
2025.03.05-04:44:22.480331      7       /dev/ttyUSB2    setproperties: setting attributes
2025.03.05-04:44:22.480399      7       /dev/ttyUSB2    setproperties: setting attributes done
2025.03.05-04:44:22.480448      4       /dev/ttyUSB2    setproperties: baudrate mismatch. ispeed ret=13 set=4098; ospeed ret=13 set=4098
2025.03.05-04:44:22.542127      7       /dev/ttyUSB2    close: flushing buffer
2025.03.05-04:44:22.542271      7       /dev/ttyUSB2    close: closing pipe read fd
2025.03.05-04:44:22.542325      7       /dev/ttyUSB2    close: closing pipe write fd
2025.03.05-04:44:22.542384      7       /dev/ttyUSB2    close: flushing with TCIOFLUSH
2025.03.05-04:44:22.542438      7       /dev/ttyUSB2    close: closing serial port
2025.03.05-04:44:22.545878      7       /dev/ttyUSB2    close: closed
```

To confirm the issue is not with the serial port itself, I created a python3 script utilizing pyserial library and it worked fine.
Once again to make sure of it again, i had a wrapper around https://github.com/imabot2/serialib, i tried that and it worked fine as well.

abdullahbaa5, 2025-03-05:
I commented out this if check for baudrate and tried the app, the port opens just fine but reading and writing does not work.

logs:
```
2025.03.05-05:14:47.778313	6	/dev/ttyUSB2	open: succeeded
2025.03.05-05:14:47.778628	7	/dev/ttyUSB2	setproperties: getting attributes
2025.03.05-05:14:47.778668	7	/dev/ttyUSB2	setproperties: setting attributes
2025.03.05-05:14:47.778688	7	/dev/ttyUSB2	setproperties: setting attributes done
2025.03.05-05:14:47.778705	4	/dev/ttyUSB2	setproperties: baudrate mismatch. ispeed ret=13 set=4098; ospeed ret=13 set=4098
2025.03.05-05:14:47.792316	2	/dev/ttyUSB2	waitformodemevent: error in modemeventthread: errno=25
2025.03.05-05:14:47.792493	2	/dev/ttyUSB2	waitformodemevent: sem_destroy: errno=25
2025.03.05-05:14:47.795480	2	/dev/ttyUSB2	waitformodemevent: error in modemeventthread: errno=25
2025.03.05-05:14:47.795527	2	/dev/ttyUSB2	waitformodemevent: sem_destroy: errno=25
2025.03.05-05:14:47.795884	2	/dev/ttyUSB2	waitformodemevent: error in modemeventthread: errno=25
2025.03.05-05:14:47.795916	2	/dev/ttyUSB2	waitformodemevent: sem_destroy: errno=25
2025.03.05-05:14:47.796163	2	/dev/ttyUSB2	waitformodemevent: error in modemeventthread: errno=25
2025.03.05-05:14:47.796209	2	/dev/ttyUSB2	waitformodemevent: sem_destroy: errno=25
2025.03.05-05:14:47.796453	2	/dev/ttyUSB2	waitformodemevent: error in modemeventthread: errno=25
// tons of the same waitformodemevent error repeating
```

jcurl, 2025-03-05:
Logs are saying wanted to set is 4098 = 010002 = B115200, so the request from C# to the library is expected. But after setting the baudrate, it says that 13 = 015 = B9600 is actually used. So setting the baudrate didn't work as expected.

if you say another implementation works, I can suggest that you can compare the two with the `strace` command looking at the differences and posting here. Perhaps we find out why there is two different behaviours.

jcurl, 2025-04-25:
Any update? If you found a solution, it would be worthwhile to post here for others that might find a similar issue.

abdofallah, 2025-04-26:
> Any update? If you found a solution, it would be worthwhile to post here for others that might find a similar issue.

I use SerialPortStream to communicate with GSM Modules such as Simcom7600g, in linux, ModemManager is installed by installed that was communicating with the port hence i was not able to use it. Where as my other program it worked sometimes and it didnt the other so i figured something else was the issue.

We can close this, it was a system/environment -issue from my side.

closed 2025-04-26 by abdofallah