jcurl/RJCP.DLL.SerialPortStream #154: Is there any way to block "Write()" until the serial port transmission is completed

ych325608, 2025-04-25:
Is there any way to block "Write()" until the serial port transmission is completed

jcurl, 2025-04-25:
Serial port protocols generally don't rely on when the buffer is flushed, but send data, and get confirmation that the data is sent by receiving new data. In this case, it's not required to wait for a write() to complete.

However, if you must ensure that the write is complete for some reason, use the flush functionality. You might want to look at the test case here: https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/master/test/SerialPortStreamNativeTest/ReceiveTransmitTest.cs#L204

ych325608, 2025-04-25:
Thank you very much for your answer! However, in my actual test, the theoretical time for sending 100,000 bytes was 100,000 *10/115200=8.68 seconds, but the actual time was about 5 seconds

[image]

ych325608, 2025-04-25:
My OS is windows10 and the.net version is 8.0
Thank you!

ych325608, 2025-04-25:
`static async Task Main(string[] args) {
    var serialPort = new SerialPortStream("COM4");
    serialPort.BaudRate = 115200;
    serialPort.Open();
    var buffer = new byte[100000];

    Stopwatch sw =Stopwatch.StartNew();
    serialPort.Write(buffer);
    serialPort.Flush();
    Console.WriteLine((sw.ElapsedTicks / (double)Stopwatch.Frequency) * 1000);

    Console.Read();
}`

jcurl, 2025-04-25:
Initially looking at the code, it appears to only wait for the memory buffers of the process to be empty, and may not wait for when the serial device itself flushes the buffers.

I'll need some time to look at the code and look for an answer. I appreciate your patience.

jcurl, 2025-04-25:
For now, could you please test if the latest version from the v2.x branch https://www.nuget.org/packages/SerialPortStream/2.4.2 works with the `Flush()` method as expected? I'm still investigating it.

jcurl, 2025-04-28:
I believe the Prolific driver is broken. I tested using the RA and TA chipsets and there is no workaround, nor any function documented by Microsoft to know when the driver buffers are empty. 

EV_TXEMPTY is raised too soon. Querying the number of bytes in the driver returns 0 (clearly wrong). FlishFileBuffers does nothing. PurgeComm with TXCLEAR returns immediately. 

If you can't adjust the protocol to get feedback when the device has received data then use the CH341 or CH343 or FTDI. 

I have no other options.

huohuaing, 2025-04-28:
Thank you very much for your answer!
我手上还有一个仪器，芯片型号是STM32H750,他直连USB的D+、D- 他写入时间更是惊人的达到了109ms，不过在更换System.IO.Ports库也是一样也是100ms。不同芯片组的逻辑差别很大？

[image]

jcurl, 2025-04-28:
It could be very helpful enabling logs to see why. Internally my library does an overlapped write, and if the driver just copies data without waiting for it to be sent, we would see this kind of behaviour.

jcurl, 2025-05-03:
Thanks for the hint that it works on the Microsoft implementation. I've compared the two and I've got it working. With the version of SerialPortStream 3.0.2 and earlier, you can do the following, and it will work:

```csharp
                var serialPort = new RJCP.IO.Ports.WinSerialPortStream(com, 115200, 8, RJCP.IO.Ports.Parity.None, IO.Ports.StopBits.One);
                serialPort.Settings.WriteTotalTimeoutMultiplier = 0;
                serialPort.Settings.WriteTotalTimeoutConstant = 0;
                serialPort.Open();

                Stopwatch sw = Stopwatch.StartNew();
                serialPort.Write(buffer, 0, buffer.Length);
                Console.WriteLine("Elapsed time: {0:f1}ms write", (sw.ElapsedTicks / (double)Stopwatch.Frequency) * 1000);
                serialPort.Flush();
                Console.WriteLine("Elapsed time: {0:f1}ms flushed", (sw.ElapsedTicks / (double)Stopwatch.Frequency) * 1000);
```

Note, that I'm creating a `WinSerialPortStream`, which exposes settings specific to Windows. The default timeout multiplier is 500ms, which turns out that for the PL2303 chipset causes problems. The Microsoft solution uses https://github.com/dotnet/runtime/blob/main/src/libraries/System.IO.Ports/src/System/IO/Ports/SerialStream.Windows.cs#L652 zero, where the default I used is 500ms to exit. Looks like the prolific driver doesn't work here properly, where the FTDI driver does.

The CH340 and the CH343 completely ignore this setting and treats it always as zero (it never returns early from a Write).

Anyway, this code works and I tested it on the various devices I have. I'll do some more testing and probably make the zero default.

jcurl, 2025-05-03:
I've pushed version 3.0.3 to NuGet. You should try that.

closed 2025-05-03 by jcurl