jcurl/RJCP.DLL.SerialPortStream #156: On Linux, lock serial ports for exclusive access using flock()

cmcqueen, 2025-06-13:
In `dll/serialunix/libnserial/openserial.c`, I notice the code can optionally do a `TIOCEXCL` lock.

```c
#if defined HAVE_TERMIOS_EXCLUSIVE
  if (ioctl(handle->fd, TIOCEXCL)) {
    nslog(handle, NSLOG_NOTICE, "open: error setting TIOCEXCL: errno=%d", errno);
  }
#endif
```

But in Linux, it's common to use a `flock()` advisory lock on serial ports. Something like

```c
  if (flock( handle->fd, LOCK_EX | LOCK_NB )) {
    nslog(handle, NSLOG_NOTICE, "open: error getting exclusive lock: errno=%d", errno);
  }
```

jcurl, 2025-10-24:
The LWN post suggests it is more complex. https://lwn.net/Articles/1041316/

cmcqueen, 2025-11-30:
Yes, you're right, there is still the legacy of UUCP-style lock files to consider.

However, long-term it seems that everyone is still moving away from UUCP-style lock files, and towards `flock()` (preferring `flock()` over `ioctl(fd, TIOCEXCL)`).