jcurl/RJCP.DLL.SerialPortStream #159: The SerialPortStream library is unable to receive data.

ms1094392787, 2025-09-08:
My serial communication device transmits approximately 100 packets of data per second, with each packet containing around 80 bytes of data, and it sends these data to the PC end at a fast rate. I established serial communication using SerialPortStream. The event that triggers the data has never been able to be waited for.
I tried subscribing to the DataReceived event, but the result was that this event would not be triggered.
I also tried to use a "While" loop within Task.Run to read the "BytesToRead" field, but it always returned 0.

jcurl, 2025-09-09:
Sorry to hear. Could you please capture some verbose logs? Very strange that it can't receive anything. Have you tried disabling flow control?

There are some free tools to monitor communications, such as sysinternals. Perhaps that could give more information on what is going wrong?

jcurl, 2025-09-20:
Adding to the printout {{Environment.GetTickCount64}}, it appears that the callback on my software occurs every 100-130ms.
```text
COM NAME?
com5
BaudRate?
115200
SerialPortStream or SerialPort?
1 or 2?
1
246926468: SerialPortStream read byte :1002 - 1002
246926562: SerialPortStream read byte :974 - 974
246926671: SerialPortStream read byte :836 - 836
246926796: SerialPortStream read byte :850 - 850
246926890: SerialPortStream read byte :822 - 822
246927015: SerialPortStream read byte :836 - 836
246927109: SerialPortStream read byte :774 - 774
246927218: SerialPortStream read byte :822 - 822
246927328: SerialPortStream read byte :774 - 774
246927421: SerialPortStream read byte :822 - 822
246927531: SerialPortStream read byte :774 - 774
246927640: SerialPortStream read byte :898 - 898
246927765: SerialPortStream read byte :836 - 836
246927875: SerialPortStream read byte :836 - 836
246927968: SerialPortStream read byte :774 - 774
246928078: SerialPortStream read byte :836 - 836
246928187: SerialPortStream read byte :836 - 836
246928296: SerialPortStream read byte :760 - 760
246928406: SerialPortStream read byte :898 - 898
246928515: SerialPortStream read byte :774 - 774
246928625: SerialPortStream read byte :836 - 836
246928718: SerialPortStream read byte :822 - 822
246928828: SerialPortStream read byte :774 - 774
246928937: SerialPortStream read byte :822 - 822
246929046: SerialPortStream read byte :836 - 836
246929156: SerialPortStream read byte :774 - 774
246929265: SerialPortStream read byte :822 - 822
246929375: SerialPortStream read byte :850 - 850
```

The reason for this is in the code https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/v2.x/code/Native/Windows/CommOverlappedIo.cs#L213-L216 which configures the timeout to be 100ms. The decision to do it this way was to reduce the CPU load when reading.

If you want to configure these, you'd need to use version 3.0.3, where the code is now given at https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/master/code/Serial/Windows/CommOverlappedIo.cs#L216-L222. i.e. you need to provide the settings with `WinNativeSettings`.

Using the SerialPortTest shows a lot of callbacks.
```text
COM NAME?
com5
BaudRate?
115200
SerialPortStream or SerialPort?
1 or 2?
2
246865703: SerialPortStream read byte :432 - 506
246865703: SerialPortStream read byte :12 - 12
246865718: SerialPortStream read byte :62 - 62
246865718: SerialPortStream read byte :12 - 12
246865718: SerialPortStream read byte :62 - 62
246865734: SerialPortStream read byte :12 - 12
246865734: SerialPortStream read byte :62 - 62
246865734: SerialPortStream read byte :12 - 12
246865734: SerialPortStream read byte :62 - 62
246865750: SerialPortStream read byte :12 - 12
246865750: SerialPortStream read byte :62 - 62
246865750: SerialPortStream read byte :12 - 12
246865765: SerialPortStream read byte :62 - 62
246865765: SerialPortStream read byte :12 - 12
246865765: SerialPortStream read byte :62 - 62
246865781: SerialPortStream read byte :12 - 12
246865781: SerialPortStream read byte :62 - 62
246865796: SerialPortStream read byte :12 - 12
246865796: SerialPortStream read byte :62 - 62
246865796: SerialPortStream read byte :12 - 12
246865796: SerialPortStream read byte :62 - 62
246865812: SerialPortStream read byte :12 - 12
246865812: SerialPortStream read byte :62 - 62
246865812: SerialPortStream read byte :12 - 12
246865828: SerialPortStream read byte :62 - 62
246865828: SerialPortStream read byte :12 - 12
246865828: SerialPortStream read byte :62 - 62
246865828: SerialPortStream read byte :12 - 12
246865843: SerialPortStream read byte :62 - 62
```

Could you please test, by reading the data when you get the callback? Reading should _never_ block on my implementation, because it just copies from an internal buffer to your buffer. It never waits for the driver if you have more than one byte in the buffer.

jcurl, 2025-09-20:
I switched the test program over to use `RJCP.SerialPortStream` version 3.0.3, and modified the class, which reads must faster:

```csharp
        //SerialPortStream sps = new SerialPortStream();
        WinSerialPortStream sps;

        public SerialPortStreamTest(string com,string baud)
        {
            sps = new WinSerialPortStream(com);
            //sps.PortName = com;
            sps.BaudRate = int.Parse(baud);
            sps.DataBits = 8;
            sps.Parity = Parity.None;
            sps.StopBits = StopBits.One;
            sps.ReadBufferSize = 4096;
            sps.DataReceived += Sps_DataReceived;
            sps.Settings.ReadIntervalTimeout = 1;
            sps.Settings.ReadTotalTimeoutConstant = 1;
            sps.Settings.ReadTotalTimeoutMultiplier = 0;
            sps.Open();
        }
```

Shows the output:
```text
COM NAME?
com5
BaudRate?
115200
SerialPortStream or SerialPort?
1 or 2?
1
SerialPortStream read byte :228; read=228
SerialPortStream read byte :138; read=138
SerialPortStream read byte :90; read=90
SerialPortStream read byte :138; read=138
SerialPortStream read byte :90; read=90
SerialPortStream read byte :152; read=152
SerialPortStream read byte :76; read=76
SerialPortStream read byte :138; read=138
SerialPortStream read byte :90; read=90
SerialPortStream read byte :152; read=152
SerialPortStream read byte :76; read=76
SerialPortStream read byte :152; read=152
SerialPortStream read byte :138; read=138
SerialPortStream read byte :90; read=90
SerialPortStream read byte :138; read=138
SerialPortStream read byte :90; read=90
SerialPortStream read byte :138; read=138
SerialPortStream read byte :90; read=90
SerialPortStream read byte :138; read=138
SerialPortStream read byte :90; read=90
SerialPortStream read byte :138; read=138
```

My software is a little bit slower, because it uses a threadpool to handle events, decoupling itself from the driver.